UofA-SPEAR/embedded #30: SWO Infrastructure

davidlenfesty, 2021-08-14:
It would be quite nice to have printf-style debugging available, so someone should write a ChibiOS "module" (i.e. makefile, header, and source) that sets up ITM channel 0 for SWO print debugging, and hooks it into the C stdlib mechanisms, so that we can use `printf` to print to debugging.

minh7a6, 2021-08-15:
Chibios already has printf style (chprintf) that utilizes usart2 on the Speedy

davidlenfesty, 2021-08-16:
AFAIK cortex-debug has support, or at least openocd will print out SWO in the terminal.

Mostly an exploratory issue, if someone looks into it and it's a ton of work we can just close it and drop this as a concept.

JackOpgenorth, 2022-05-14:
Closing as part of cleanup since this is not a priority

closed 2022-05-14 by JackOpgenorth