UofA-SPEAR/embedded #38: Revamp Onboarding

davidlenfesty, 2021-09-07:
As it exists, the onboarding is pretty terrible. (hrm... I wonder who wrote it)

I believe it should be re-written to actually be thorough, and not assume people will know where to look. It should function more like a tutorial than a list of tasks to finish, because this will be most people's first step into embedded.

The task itself should be similar, but updated for the state we are in now. So it should be reading some sensor value from the ADC, and then pushing it out over CAN.

This probably depends on #36 and  #37  getting done.

davidlenfesty, 2021-10-12:
Spent an hour on this, got a decent ways. The work (in terms of code) for this is in `david/onboarding`, that won't be the permanent place for it, just wanted to save it somewhere other than my computer.

The onboarding document itself is in much rougher shape than the code, but I'll update it in the wiki anyways, as it's probably more informative in it's current state than the existing onboarding. [Wiki Onboarding](https://github.com/UofA-SPEAR/embedded/wiki/Onboarding-Task)

I want to add another task in onboarding, to listen for incoming Heartbeat messages and then blink according to the number of nodes it sees. That way there's a simpler thing that people can do and they can see the effects of what they wrote by just plugging the board into the rover. I'll probably slightly restructure this so that this task comes before the ADC task, although I believe the ADC is the meat of the whole onboarding thing, as it takes people into proper embedded concepts and is a more practical example of using ChibiOS.

JackOpgenorth, 2022-05-14:
Onboarding is better than it was before, but some more improvements would be nice, so this issue is staying open