UofA-SPEAR/embedded #82: CAN - USB adapters

JackOpgenorth, 2022-09-17:
Having these would be nice since we wouldn't have to ssh into the tx2 every time we wanna see CAN

JackOpgenorth, 2022-09-24:
We found one that looks promising [here](https://www.aliexpress.com/item/1005003746105255.html?spm=a2g0o.productlist.0.0.34f23e9bPfOiW0&algo_pvid=2a6770a8-da62-40b7-a8a5-c1342cf264f4&algo_exp_id=2a6770a8-da62-40b7-a8a5-c1342cf264f4-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000027291719020%22%7D&pdp_npi=2%40dis%21CAD%2127.73%2127.73%21%21%21%21%21%402101e9d216634471781198184ec35c%2112000027291719020%21sea&curPageLogUid=5jldTdn2YU0G). 5 should be plenty

nic3-14159, 2023-01-15:
SLCAN on the SPEEDYs works now, so this issue is complete

closed 2023-01-15 by nic3-14159